thorrak/tiltbridge #6: Add support for Brewfather API KEY / Logging ID

tgangso, 2019-04-05:
Please add support for Brewfather API KEY / Logging ID so i can recommend this to my users.

myndflyte, 2019-06-04:
I would also like tiltbridge integration with brewfather.

omgitsads, 2019-07-19:
For logging to Brewfather, you'd want to use the [Custom Stream integration](https://docs.brewfather.app/integrations/custom-stream), rather than the existing Tilt integration. The payload is almost identical to the payload for Brewer's Friend.

I have a fork here: https://github.com/omgitsads/tiltbridge/tree/brewfather that just switches out the URL, which is working for me. I'm going to try looking at getting both brewer's friend and brewfather running at the same time, but was running into the partition limits mentioned in https://github.com/thorrak/tiltbridge/blob/master/docs/source/development.rst.

I think `void dataSendHandler::send_to_brewers_friend` could probably be adapted to take an arbitrary URL and JSON key, but think the bigger issue is adding the additional HTML for configuration would push it over the partition limits.

tgangso, 2019-07-20:
Great work @omgitsads thanks. Was not avare of the partition limits. Hopefully it can be worked out. I will ask some to test out your Brewfather version.

brookwarner, 2019-11-26:
I'm trying to test it and can't get it working but my knowledge of the esp32 development is very basic. 

i can get master running fine, but can't get the brewfather fork to even run, let alone send to BF

thorrak, 2019-11-26:
The (official) TiltBridge support for BrewFather hasn't been created/released yet, unfortunately. For an update on what is going on more broadly with the TiltBridge, take a look at the [HBT post here](https://www.homebrewtalk.com/forum/threads/tiltbridge-tilt-to-wifi-device-for-tilt-hydrometer.664368/page-2#post-8715264). I'm hoping that once my _other_ project is released, the same hardware can be reused for TiltBridge thus resolving that source of (significant) frustration.

The one thing that post _doesn't_ talk about is the problem that @omgitsads noted - the partition limits. The hope I have there is that although it feels like the ESP32 is a mature platform, in comparison to the ESP8266 (or Arduino, or...) it is actually quite new. A number of libraries - Bluetooth chief amongst them - weren't optimized either for space or speed as of a year and a half back when I first started TiltBridge. My hope is that in the intervening period optimization has taken place to reduce the footprint of the firmware.

corbinstreehouse, 2019-12-07:
FWIW, I added this support and did a pull request. It doesn't seem to hit partition limits yet.

tgangso, 2019-12-11:
Fantastic

thorrak, 2020-12-22:
This was added awhile back and Brewfather is now one of the core services I'm testing against. Closing.

closed 2020-12-22 by thorrak