thorrak/tiltbridge #7: Broken flashing for "TTGO" OLED board

bjornsol, 2019-05-21:
Flashing the ["TTGO" OLED board](https://www.aliexpress.com/item/Lolin-ESP32-OLED-V2-0-Pro-ESP32-OLED-wemos-pour-Arduino-ESP32-OLED-WiFi-Modules-Bluetooth/32822105291.html) listed in the documentation with esptool or Fermentrack does not work (or my boards may be duds). 

Bought two boards, result:

Flashed one board by using fermentrack, now the board has no activity on the screen and the status led is glowing constantly with a bright blink each second. This behaviour continues after ereasing the flash and trying to reflash with esptool.

Flashed the other board with esptool there is no activity on the screen and the status led is glowing constantly, but without any blinking.

All flashing processes returned success with no timeout or other issues.

thorrak, 2019-05-21:
Do you happen to be located in the US?

bjornsol, 2019-05-21:
Nope, Northern Europe

thorrak, 2019-05-21:
If you’re on either HomeBrewTalk or Reddit, do you mind sending a private message to Thorrak or /u/lordfili respectively? Depending on shipping cost, I’d like to swap boards with you to try to figure out what is going on.

bjornsol, 2019-05-21:
Check you messages on HomeBrewTalk

thorrak, 2020-12-22:
This should be resolved using Brewflasher. Closing.

closed 2020-12-22 by thorrak