thorrak/tiltbridge #12: Missing monitor_speed in PlatformIO.ini

lbussy, 2019-12-10:
There's no monitor_speed in the ini file to correspond to the Serial.begin() statement.

As an aside, any reason to disable Serial printing?  You know, aside from:

```
DATA:    [==        ]  21.2% (used 69628 bytes from 327680 bytes)
PROGRAM: [==========]  98.2% (used 1931073 bytes from 1966080 bytes)
```
 :)

lbussy, 2019-12-12:
By the way, I learned you can use an "upload speed" as well to make things faster.  The settings I am currently using are:

```
[common_env_data]
upload_speed = 460800
monitor_speed = 74880
```

You can then reference them in your board's section:

```
[env:lolin32_lcd_ssd1306]
upload_speed = ${common_env_data.upload_speed}
monitor_speed = ${common_env_data.monitor_speed}
```

thorrak, 2020-12-22:
This was added awhile back - closing

closed 2020-12-22 by thorrak