thorrak/tiltbridge #18: connect to hidden ssid

monty124, 2020-02-11:
testing tiltbridge on a ESP32 OLED and I cannot seem to get it to connect to a hidden ssid wifi connection.
In addition once i do connect it to a broadcast ssid, i cant connect directly using the ip displayed on the oled screen and on a power cycle nothing is retained, it goes back to the tiltbridgeap configure screen.
are there any logs we can look at to try and debug?

thorrak, 2020-02-11:
There are not, unfortunately.

monty124, 2020-02-11:
ok thanks for replying, should it hold the config on power cycling? any tips on where i can look to try and work out what is going on? i can see some refrences to serial output in the code but it does not look like much

lbussy, 2021-01-10:
I tested this creating a soft AP with a hidden SSID as follows:

``` cpp
#include <ESP8266WiFi.h>
#include <Arduino.h>

const char *ssid = "softap";
const char *password = "password";

void setup()
{
  WiFi.softAP(ssid, password, 1, 1); 
}

void loop() { ; }
```

I am unable to reproduce this isse.  Can you provide additional information?

monty124, 2021-01-11:
oh wow this was indeed a while ago! 
I couldn't get this working, i was trying to connect to a hidden ssid from the built in web interface but it would not save
I ended up abandoning this as I could not get it to work, I'll close this comment and if i get a change to revisit and have issues again I will log an issue
thanks for taking a look anyway.

closed 2021-01-11 by monty124