thorrak/tiltbridge #19: Special characters in WPA2 PSK Passwords

skroz1, 2020-03-08:
It appears that certain special characters are not being escaped correctly in the MPA2 PSK passphrase.  In testing, the Tiltbridge was not able to join a 2.4 GHz wifi network when the PSK contained a $ character.  Removing this character resolved the issue.

lbussy, 2021-01-10:
I created a soft AP with which to test using the following:

``` cpp
#include <ESP8266WiFi.h>
#include <Arduino.h>

const char *ssid = "soft-access-point";
// const char *password = "$password";
const char *password = "pa$$word";

void setup()
{
  WiFi.softAP(ssid, password);
}

void loop() { ; }
```

I tested connecting TiltBridge to the AP with passwords `$password` and `pa$$word` and both worked as expected.  I am unable to reproduce.

lbussy, 2021-01-13:
Tested ok during a review.   Closing.  Can re-open with additional detail.

closed 2021-01-13 by lbussy