thorrak/tiltbridge #20: Trouble installing

sandbreed, 2020-03-19:
I have a MacBook with OS 10.14.6 and am trying to install to an ESP32 OLED board. The Mac has Python 2.7 but I tried with 3.8 as well. I used "pip install esptool" to install esptool and used "pip show -f esptool" to confirm it was installed. Using "esptool.py" results in "command not found." Using "/Users/USER/Library/Python/2.7/bin/esptool.py" provides this: 

usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: too few arguments

Using "esptool.py --chip esp32 --before default_reset --after hard_reset write_flash 0x10000 firmware.bin 0x8000 partitions.bin 0x3D1000 spiffs.bin" results in "command not found." I am not sure what to do next. 

Also, do I move piffs.bin, firmware.bin, and partitions.bin to the same folder as esptool.py? I have tried with the files there and left in Downloads but need to confirm. I would appreciate any help.

thorrak, 2020-03-19:
Make your life easier. Use BrewFlasher. :)

sandbreed, 2020-03-19:
I wish I knew about that hours ago. Everything seemed to work but the board does not boot, or at least the wireless network TiltBridgeAP does not show on my Mac or phone. When I plug it into a wall outlet, a red light turns on but turns off in about 1 second without anything on the screen.

thorrak, 2020-03-19:
Sorry about that - I need to do a better job of promoting it. :(

Which firmware did you flash - the regular OLED one or the beta OLED one?

sandbreed, 2020-03-19:
The regular version

sandbreed, 2020-03-19:
I ended up installing the beta and everything seems okay so far. I really appreciate the help and BrewFlasher!

sandbreed, 2020-03-21:
Apparently I'm just not supposed to use this. The beta version loaded and I setup Google Sheets. The TiltBridge logged one reading but since then has not logged another. I tried repeating the entire process with the same result.

thorrak, 2020-03-21:
You are not alone in having issues with google sheets. It's one of the two reasons the firmware is still tagged Beta. :(

Sit tight - With the whole "no bars and restaurants" thing, I'm likely going to be focusing on fixing this over this weekend.

sandbreed, 2020-03-21:
Gotcha. I appreciate what you have done already.

thorrak, 2020-12-22:
This should be fixed with a prior release - closing the issue

closed 2020-12-22 by thorrak