thorrak/tiltbridge #37: Lolin TFT screen

speto, 2020-11-09:
Hi there, great work on TiltBridge 🥇 
I'm not sure if this is the right place, but Lolin TFT screen mentioned in hardware section https://github.com/thorrak/tiltbridge/blob/master/docs/source/hardware.rst#lolin-d32-pro--tft seems to be no longer available for order.
Any ideas for a suitable screen replacement?

thorrak, 2020-11-09:
UGH. Lolin's stocking has been frustratingly sporadic - both on the D32 Pros and the screens. I love their components for TiltBridge since it reduces the build to snapping a single cable into two sockets, but I wish that they were reliably in stock.

Lolin does tend to vacillate between having these show as temporarily out of stock vs. unavailable so I wouldn't be surprised if they restocked in the near future.

If you want to get something built, the firmware still supports the [Heltec OLED boards](https://www.aliexpress.com/item/32958037260.html). There may be features implemented in the future that leverage the touchscreen on the Lolin TFTs which the Heltec boards wouldn't support, obviously, but it will work to bridge a Tilt to WiFi.

thorrak, 2020-11-11:
Here's what I got back from LoLin when I asked them on AliExpress:

[image]

...so it sounds like it will be coming back in stock at some point.

thorrak, 2020-11-23:
THey're back: https://www.aliexpress.com/item/32919729730.html

Appears to be the 1.0.0 version. I still need to get the code in for rotating the screen for the 1.1.0 version, so 1.0.0 is good. ;)

speto, 2020-11-23:
Thanks for information. 
I've seen that on Friday. Ordered one just as I saw that. The board with cable should be on their way already, I ordered them earlier. Now I can't way to put it together. 
I really like the simplicity of this solution over Raspberry or phone/tablet. 

I hope that information in this issue may help somebody else in the future if screens will be out of stock again.
For now I suppose we can close it.

closed 2020-11-23 by speto