thorrak/tiltbridge #48: Local Report JSON: mDNS ID Incorrect

lbussy, 2020-12-27:
Regardless of the mDNS name of the controller, the JSON still lists the mDNS name as "tiltbridge."

It would be ideal/cleaner if the report listed a service name ('api_name'?) as well as the mDNS ID.

pletch, 2020-12-28:
Can you clarify what 'report' you are looking at?  Is this from JSON payload sent to one of the configurable endpoints?

lbussy, 2020-12-28:
Yes. I'm using the Local Target push.

pletch, 2020-12-28:
That's odd.  The local target push JSON payload includes the correct mDNS ID.  For example, here is the debugging output of the send to the Local Target Push endpoint.   

Data to send: {"mdns_id":"tiltbridge2","tilts":{"Black":{"color":"Black","fwVersion":0,"gravity":"1.0520","gsheets_name":"","high_resolution":false,"sends_battery":true,"temp":"71.0","tempUnit":"F","weeks_on_battery":25}}}

Unfortunately, I don't have an actual Local Target Push endpoint set up to receive to troubleshoot further. However, the payload now includes new data that wasn't there before prior to v.2.2.  Perhaps something needs to be updated on the receiving side.  This may also relate to the issues that one of the users reported on HBT regarding difficulty receiving in Fermentrack.

@thorrak will have to chime in on this one.

lbussy, 2020-12-28:
> However, the payload now includes new data that wasn't there before prior to v.2.2. Perhaps something needs to be updated on the receiving side.

Yep, I was part of that discussion. 👍 

I'll give it another go after my morning calls.  I need to review this and #51 so I'll mess with them at the same time.

If you need a very basic endpoint with which to test (and you have PHP) you can use [this](https://github.com/brewpi-remix/brewpi-www-rmx/blob/master/brewpi-api.php) file.  Turn on debug on line 20 and comment out around 62 where it posts into BPR.

lbussy, 2020-12-28:
This was Lee being a dumbass.  I forgot (even after telling that guy on HBT) that the settings (including the mDNS name) would reset when the update was flashed.

Closing.

closed 2020-12-28 by lbussy