thorrak/tiltbridge #51: Tilt Version Fails to Reset

lbussy, 2020-12-27:
Scenario:

- Two Tilts of the same color
- One Pro
- One < Pro
- Log for a time with the Pro
- Turn the Pro off
- Turn the non-Pro on
- Observe the readings are off by a factor of 10

I suspect the version checking is only done once.  This could be done each time (cheap enough processing remaining) or at the least, it should be done on a timer.

Also, gravity smoothing gets very weird.  It might be cleanest to track on a MAC address and reset the buffer if the MAC changes.

pletch, 2020-12-28:
I assume you must be turning the non-Pro on before the Pro actually expires and disappears from the dashboard?

lbussy, 2020-12-28:
> I assume you must be turning the non-Pro on before the Pro actually expires and disappears from the dashboard?

Not sure, what's the timeout?

pletch, 2020-12-28:
5 minutes

lbussy, 2020-12-28:
I'm an edge case!   (We always knew that.)

I can confirm it resets as you intended and that you have also added some additional logic to address my "color issues."  This should go away completely when John and I have a chance to swap our Tilt Pros.

Closing.

commit d616e5b6 (2020-12-28): Merge pull request #52 from pletch/tweaks

Handle unexpired swap from Pro to non-Pro (Fixes #51)
closed 2020-12-28 by lbussy