thorrak/tiltbridge #87: Add Google Sheets Link

lbussy, 2021-01-11:
Add a link to the Google Sheet in the main view.

lbussy, 2021-01-13:
Thinking about this more - to do this I'd need to add the Google Script URL to the JSON report.  That's a higher lift than I thought originally and now I'm considering whether I want to add this link or not.

thorrak, 2021-01-14:
I don't think we currently collect enough information to be able to do this, as the mapping between Sheet Name and Sheet URL is inside the TiltBridge Google Sheet "App". We can build an endpoint to redirect to the google script URL -- but that's different from the google sheet itself.

Proposing we tag this as `wontfix` and close.

lbussy, 2021-01-14:
I reject your reality and insert my own!  Let's leave it open and if it's the last stinky one with no ideas, close it out?

lbussy, 2021-01-18:
Working on this in `gsheet_link`.

PR #135 (2021-01-21): Gsheets link
Closes #87.
closed 2021-01-21 by lbussy