thorrak/tiltbridge #96: Scan already in progress - explicitly stopping

lbussy, 2021-01-13:
I did not notice if BLE updated or not, but I get this now:

```
     30678 V: Scan already in progress - explicitly stopping.
     30784 V: Scan already in progress - explicitly stopping.
     30888 V: Scan already in progress - explicitly stopping.
     30993 V: Scan already in progress - explicitly stopping.
     31098 V: Scan already in progress - explicitly stopping.
```

This will take some digging I imagine.

pletch, 2021-01-13:
I noticed this issue yesterday and rolled back to a snapshot to confirm that by this commit to Nimble-Arduino [](https://github.com/h2zero/NimBLE-Arduino/commit/413e64640946195bf5f68c384279a615aefe8600l).   

Are you going to tackle or would you like me to look into it?

pletch, 2021-01-13:
I think there might be a bug in NimBLE as the return code from starting a scan is now always zero.  Submitted [issue](https://github.com/h2zero/NimBLE-Arduino/issues/175) on NimBLE github.

lbussy, 2021-01-13:
> Are you going to tackle or would you like me to look into it?

I have not had a chance to look yet - threw the issue up here so I would not forget.  If you have time, great.  I don't forsee time today but if I get back into it I'll assign the issues I am working on to myself.

lbussy, 2021-01-13:
I see he's fixed it, should just need testing?

PR #102 (2021-01-13): Tiltscan update
Closes #96
closed 2021-01-13 by lbussy