thorrak/tiltbridge #128: Google Sheets endpoint sporadically posts same data twice or three times

pletch, 2021-01-18:
Google sheets endpoint always initially posts data twice and will sometimes randomly post multiple data points during the same sending event.    See examples below.  This is something associated with Send_with_redirects as I did a test by switching to using the [HTTPSRedirect library](https://github.com/electronicsguy/ESP8266/tree/master/HTTPSRedirect) (which has its own issues....) temporarily and it does not exhibit these anomalies.

**Initial Startup Example**
[image]

**Random multi-POST of the same data point.**
[image]

thorrak, 2021-01-18:
Interestingly enough, I think the missing functionality may have been added to the standard HTTPClient library [late last year](https://github.com/espressif/arduino-esp32/commit/ee88c42c3b5e87a2fc09303f9e82dd8172e99990). 

I wouldn't mess with this just yet as I know @lbussy was looking at that section of code, but it might be possible to strip `send_with_redirects` out entirely as a result of this change.

lbussy, 2021-01-18:
Yep, working on this in addition to the others.

lbussy, 2021-01-19:
I am 91.73% sure this was related to the 301 handlers which is no longer needed with the new code I am working on.

lbussy, 2021-01-19:
Unable to reproduce since refactor - closing as fixed with #132.

closed 2021-01-19 by lbussy