thorrak/tiltbridge #129: Use of Google Sheets endpoint causes excessive fragmentation

pletch, 2021-01-18:
When using the Google Sheets endpoint, after some period of time, fragmentation will increase to 50-60% and never resettle back down in the 20-30% range that is typical.  It is unknown whether this is recent behavior or was an issue prior to recent changes.

thorrak, 2021-01-18:
I'm wondering if the changes made available due to the addition of native redirect support would actually allow us to just instantiate a single HTTPClient instance and leave it in memory (thus reducing fragmentation). Might be worth trying.

The HTTPS implementation TiltBridge has currently was born over the course of a year and a half out of frustration with the memory limitations of the default BLE stack and the standard HTTPS client. To put it bluntly: given the issues it had initially, I'm shocked I was able to get it working at all. Given the additional maturity of the stack since it was implemented, however, There's definitely quite a bit of room for improvement.

lbussy, 2021-01-19:
This seems to be related to scanning at the same time as a POST is happening.  I have what I think is a fix in this branch I am working on.

thorrak, 2021-01-19:
> This seems to be related to scanning at the same time as a POST is happening. I have what I think is a fix in this branch I am working on.

This shouldn't have been the case, as we should have waited for a scan to stop before posting anything to google sheets. Regardless - if we still experience it after you're done, I have some thoughts.

lbussy, 2021-01-19:
It still pops up to ~50% on occasion but will drop back down.  As soon as I do the work I intended to do with this branch before re-writing everything, I'll get it pushed and you can have a look.

The behavior I saw was `wait_until_scan_complete()` without the `NimBLEDevice::deinit()` caused excessive fragmentation.  You had commented out the `deinit()` based on [an issue](https://github.com/h2zero/NimBLE-Arduino/issues/23) you raised.  That issue seems resolved so I have re-implemented the `deinit()`.

lbussy, 2021-01-19:
Definitely seems much improved with #132.  Will need further testing in regression but marking this as closed.

closed 2021-01-19 by lbussy