thorrak/tiltbridge #158: AP on channel 3

lbussy, 2021-02-06:
Apparently, the AP is coming up on channel 3 and is set at 40MHz.  The net effect is that it kills home APs on channels 1 and 6.

thorrak, 2021-02-06:
Users just need to reconfigure their networks to accommodate tiltbridge.

lbussy, 2021-02-06:
Need:

```
    // if true, enable autoreconnecting
    void          setWiFiAutoReconnect(boolean enabled);

    // set ap channel
    void          setWiFiAPChannel(int32_t channel);

    // set the country code for wifi settings, CN
    void          setCountry(String cc);
```

Also need to explore [`esp_wifi_set_bandwidth()`](https://docs.espressif.com/projects/esp-idf/en/v4.0/api-reference/network/esp_wifi.html#_CPPv422esp_wifi_set_bandwidth16wifi_interface_t16wifi_bandwidth_t).

lbussy, 2021-02-07:
Here's the issue:

[image]

lbussy, 2021-02-09:
Either the AP is on channel 3 with a 40 MHz bandwidth or is on channels 1 and 5 (as my analyzer shows) with 20 MHz each.  I can't tell with my current tools.  All I can tell is there is no currently working method to get it to channel 1 and 20 MHz.  I am going to have to raise an issue on the WiFiManager repo when I get my sh*t together.

Also, I should probably try a straight AP example without the lib.

lbussy, 2021-02-10:
Submitted [#1209](https://github.com/tzapu/WiFiManager/issues/1209) in the WiFiManager repo.

PR #159 (2021-02-12): Fix DHCP name chane and bandwith on portal
Closes #158, also handles the DHCP name issue (albeit in a not very glamorous way).
closed 2021-02-12 by lbussy