thorrak/tiltbridge #167: Null Brew Link

SillyHats, 2021-03-15:
When using tilt bridge to send data only to google sheets there is a link to the beer name (in this case 3rd) that goes to /null resulting in a 404 error

[image]

[image]

lbussy, 2021-03-15:
I suspect then that the URL is malformed?  If you can watch in the serial monitor, you will see the transaction.  Aside from that, the link will not be correct till after the sheet sends/receives once.

SillyHats, 2021-03-15:
Thanks! You are correct. 

That text links to the google sheet after the second data point is logged.

closed 2021-03-15 by SillyHats

lbussy, 2021-03-15:
Good deal.

I should have put a better link in there - I was thrilled to have the damned workflow work, however, and I forgot. :)

SillyHats, 2021-03-18:
Somebody needs to be the [DAU](https://www.acronymfinder.com/D%C3%BCmmster-Anzunehmender-User-(German%3a-Dumbest-Assumable-User%3b-play-on-words%2c-see-GAU)-(DAU).html) to find such bugs.

thorrak, 2021-03-18:
> Somebody needs to be the [DAU](https://www.acronymfinder.com/D%C3%BCmmster-Anzunehmender-User-(German%3a-Dumbest-Assumable-User%3b-play-on-words%2c-see-GAU)-(DAU).html) to find such bugs.

Normally I can be trusted to act in that role... 😆