thorrak/tiltbridge #179: Add Grainfather support

teanooki, 2021-04-05:
It would be great having the ability to send data to Grainfather. I managed to communicate with their API on the console:

- Add Tilt device under Equipment
- Copy the individual url you can find by pressing the information button. Example: https://community.grainfather.com/iot/XXXX-XXXX/tilt
- Send JSON in POST body `{"SG":1.102, "Temp":23.9, "Unit": "C"}`. Once per 15 minutes,

**Example**
`curl -H 'Content-Type: application/json' -H 'Accept: application/json' --data '{"SG":1.102, "Temp":23.9, "Unit":"C"}' -X POST https://community.grainfather.com/iot/XXXX-XXXX/tilt`

Result:
<code>
HTTP/2 201
{"result":"success"}
</code>

Error:
<code>
HTTP/2 429
{
    "message": "Too Many Attempts."
}
</code>

**Feature request**
- New option to add an individual URL for each Tilt color in the logging settings
- Send data as described to Grainfather once per 15 minutes

**Data received in Grainfather**
[image]

thorrak, 2021-04-05:
I already have Grainfather as a supported target in [Fermentrack](https://github.com/thorrak/fermentrack), it shouldn't be an issue to get this added to TiltBridge as well.

thorrak, 2022-01-13:
This is now merged to master

PR #182 (2021-04-06): Devel
Resolves: #179
closed 2022-01-13 by thorrak