thorrak/tiltbridge #194: Home Assistant MQTT discovery is not working

ddyeakley, 2021-04-21:
Home Assistant MQTT discovery is not working,   The sensors do not show up on the MQTT Integration.    While the config topic is being generating at 'homeassistant/sensor/tiltbridge_tilt_BlackT/config'  it appears with out a uniq_id the sensor is not discovered.  

The following does not work:

```
{
  "dev_cla": "temperature",
  "unit_of_meas": "°F",
  "ic": "mdi:thermometer",
  "stat_t": "tiltbridge/tilt_Black",
  "name": "Tilt Temperature - Black",
  "val_tpl": "{{value_json.Temp}}",
}
```

The following works:

```
{
  "dev_cla": "temperature",
  "unit_of_meas": "°F",
  "ic": "mdi:thermometer",
  "stat_t": "tiltbridge/tilt_Black",
  "name": "Tilt Temperature - Black",
  "val_tpl": "{{value_json.Temp}}",
  "uniq_id": "tiltbridge_tilt_black"
}
```

ddyeakley, 2021-04-21:
So actually,  sensors DO show under Configuration | Entities.     But they do not show up on the MQTT integration under Configuration | Integrations | MQTT.

thorrak, 2021-04-22:
Unfortunately, MQTT/HA isn't something that either @lbussy or I use, and as such this isn't something we can actively diagnose/repair.

Is the solution you are recommending adding a `"uniq_id": "tiltbridge_tilt_black"` key to each message being pushed for both the temp and gravity config topics?

@pletch - If you're around, this sounds like it would be a 1 (well, 2) line tweak that I'm happy to make - any reason you can think of that this would screw something up in HA?

pletch, 2021-04-22:
I don't actually use MQTT discovery but I can test this after I hack together another tiltbridge.  I think we implemented the config payload to match the sensor with multiple values example in the Home Assistant docs ([here](https://www.home-assistant.io/docs/mqtt/discovery/)). However, the first paragraph on the docs page does imply that a uniq_id should be present despite the absence in the example.  I can test and ensure there are no unwanted effects from adding this to the config payload for both the temperature and gravity sensors.

TaplistIOTeam, 2021-05-04:
I was cruising through open issues and found this. I use HA/MQTT - do y'all still need help debugging or verifying?

pletch, 2021-05-04:
If you have a tiltbridge on hand and can test MQTT discovery, that would be great. It may be a little while before i can get to it.

TaplistIOTeam, 2021-05-04:
No prob! Yep, I've got all of the above. Will give it a shot tonight.

lbussy, 2021-08-08:
> Unfortunately, MQTT/HA isn't something that either @lbussy or I use

Except I do ... what I don't do is keep an eye on issues. 😛 

I have HA anyway, I'll see if I can poke at it unless MIke beats me to it.

thorrak, 2021-08-08:
You didn't back in April. ;)

thorrak, 2022-01-11:
@ddyeakley / @TaplistIOTeam - Not sure if one of you guys can test (or has tested?) this, but I'm releasing a 1.0.3 beta here later that will include this change.

commit 2dd52aba (2022-01-11): Add uniq_id to MQTT payload (Fixes #194)
commit 769be5cb (2022-01-11): Merge pull request #202 from thorrak/mqtt_uniq_id

Add uniq_id to MQTT payload (Fixes #194)
closed 2022-01-13 by thorrak