thorrak/tiltbridge #195: Can't get the Tilt discovered when in a stainless steel conical fermentor

octave, 2021-05-24:
Hi,

I tried two different ESP32 boards, including this one with an external antenna: https://www.amazon.com/dp/B07FYWFH4C

But unfortunately both don't detect the Tilt as soon as it's inside my Spike CF5 stainless steel conical fermentor, even when rubbing it on the fermentor (works fine outside btw). My iPhone 12 manages to detect it no problem when in the fermentor.

Any tips or solution?

Thank you!

kegtron, 2021-07-25:
Placing a Tilt in a stainless conical fermentor is an extremely challenging RF environment. It's a wonder your iPhone is able to connect. It most certainly has a higher-performance Bluetooth radio vs. the $3 ESP32 chip which would seem to be just enough to get it to work on your phone.

On your iPhone install the LightBlue app which will give you a live view of the BLE signal strength to the Tilt. Methodically move your phone around the fermentor and see where the signal strength is highest. This will likely be near a non-metallic seam (e.g. the gasket on the lid). 

Once you've found the best signal location, place your ESP32 board in that same area. Try rotating the ESP32 in multiple orientations. Maybe, just maybe you'll get a connection.

In the likely event that this doesn't work you could try the TiltPro which has a higher-power transmitter.

Good luck!

lbussy, 2021-07-25:
> In the likely event that this doesn't work you could try the TiltPro which has a higher-power transmitter.

I can confirm the Tilt Pro is the answer to everything from bad connectivity to chronic halitosis.  I love mine (the Tilt, not the halitosis.)

The Tilt app will show the signal strength - it shows as a negative number like `-56dBm`.  A higher number is better, e.g. `-56` is better than `-75`.

Here's mine showing `-53dBm` all the way across the house:

[image]

thorrak, 2022-01-11:
Closing this, as there's really not a fix that TiltBridge can implement (software-wise, at least!). Some ESP32 boards have the ability to connect an external antenna -- That might help, but that's about the only solution I can think of here.

closed 2022-01-11 by thorrak