thorrak/tiltbridge #198: MQTT keepalive is being multiplied by 1000 (so, 30000 instead of 30)

bluemodena, 2021-12-24:
I think there is a bug in the MQTT code where the keep alive is being multiplied by 1000.  The value should be in seconds for eclipse-mosquito client.   All my other MQTT enable devices use seconds.

1640362453: New client connected from XXX.XXX.X.XXX as GarageMonitor (p2, c1, k15, u'mqtt-user').
1640362453: New client connected from XXX.XXX.X.XXX as Freezer (p2, c1, k30, u'mqtt-user').
1640362458: New client connected from XXX.XXX.X.XXX as EspressoClient (p2, c1, k30, u'mqtt-user').
1640362460: New client connected from XXX.XXX.X.XXX as ChristmasHouseLights-TasmotaPlug (p2, c1, k30, u'mqtt-user').
1640362464: New client connected from XXX.XXX.X.XXX as CrawlSpaceMonitor (p2, c1, k15, u'mqtt-user').
1640362467: New client connected from XXX.XXX.X.XXX as EntranceLight (p2, c1, k30, u'mqtt-user').

1640293979: New client connected from XXX.XXX.X.XXX as tiltbridge (p2, c1, k30000, u'mqtt-user').

https://github.com/thorrak/tiltbridge/blob/master/src/sendData.cpp (line#388)
        mqttClient.setKeepAlive(config.mqttPushEvery * 1000);

thorrak, 2021-12-24:
What is the impact of this?

thorrak, 2021-12-24:
In doing some research, it looks like this deviates from the standard:
https://docs.oasis-open.org/mqtt/mqtt/v5.0/os/mqtt-v5.0-os.html#_Toc3901045

bluemodena, 2021-12-24:
@thorrak, it is causing the client to disconnect.   I'm still using mqtt_v3.

https://mosquitto.org/man/mosquitto-conf-5.html
        If an MQTT v3.1.1 or v3.1 client specifies a keepalive time greater than max_keepalive they will be sent a
        CONNACK message with the "identifier rejected" reason code, and disconnected.

In my solution, I have the tiltbridge reading from TiltHydrometers, which then pushes everything to my Mosquitto server (https://hub.docker.com/_/eclipse-mosquitto/) which them publishes to my craftbeerpi4 (https://github.com/avollkopf/craftbeerpi4)

All that is needed here is to remove the '* 1000' from
           https://github.com/thorrak/tiltbridge/blob/master/src/sendData.cpp (line#388)
                      mqttClient.setKeepAlive(config.mqttPushEvery * 1000);

thorrak, 2022-01-11:
@bluemodena - I managed to locate some hardware here, and am trying to get an update for (amongst other things) this issue. If I upload a beta firmware, would you be willing to test it?

bluemodena, 2022-01-11:
@thorrak Yes, I would be happy to. While I am familiar with MQTT, I'm not that familiar with flashing ESP32s... Would this be available for testing using BrewFlasher (as this is what I originally used)?

lbussy, 2022-01-11:
Yes:

```
Updated TiltBridge - BETA - TFT_ESPI - TTGO USB-C to 1.0.3-Beta
Updated TiltBridge - BETA - LCD_SSD1306 - OLED to 1.0.3-Beta
Updated TiltBridge - BETA - D32_Pro_TFT - TFT to 1.0.3-Beta
Updated TiltBridge - ALPHA - SillyHats - TTGO USB-C to 1.0.3
```

thorrak, 2022-01-11:
I edited @lbussy 's post to remove the SIllyhats one, but the other 3 beta versions are up-to-date and include this fix

closed 2022-01-13 by thorrak

bluemodena, 2022-01-14:
I have a lot going on at work this week, so I'm late the the party, but the change looks good.  Thanks.

commit ac7051fb (2022-01-11): Change MQTT keep alive to seconds (Fixes #198)