thorrak/tiltbridge #199: Unable to rotate/invert display

jakemastera, 2022-01-07:
I recently flashed TiltBridge - v1.0.2 - OLED and when selecting the checkbox for "Invert TFT Display" under Settings>General Settings nothing on the TiltBridge changes. Power cycling the device from the webpage and pulling power both don't force a fix.  Any ideas?

thorrak, 2022-01-07:
The OLED screen is not a TFT display, so "invert" doesn't do anything. I'll take a look to see if the OLED library might have an analogous function, but I don't recall there being one.

lbussy, 2022-01-07:
Try re-flashing and check the "erase" option and see if that helps allow it to save.

lbussy, 2022-01-07:
Or, what John said.  :)

jakemastera, 2022-01-07:
> The OLED screen is not a TFT display, so "invert" doesn't do anything. I'll take a look to see if the OLED library might have an analogous function, but I don't recall there being one.

I thought that might be it, I did try a few other firmware versions but they all had the same issue - thanks for the reply!

jakemastera, 2022-01-07:
> Try re-flashing and check the "erase" option and see if that helps allow it to save.

I did try that with a few different firmware versions that were available through BrewFlasher but they all seemed to have the same effect.

thorrak, 2022-01-07:
Upon investigating, as I expected the code to actually do something with the "Invert TFT" option checked is [only enabled](https://github.com/thorrak/tiltbridge/blob/f0ed2cfd0ad45a1b3f4dcac845e7de1732db4564/src/bridge_lcd.cpp#L101) for the TFT-style displays.

There IS an analogue in the OLED library, however:
```
// Turn the display upside down
void flipScreenVertically();
```

I don't have Tilts, an OLED tiltbridge, or an environment that is set up to compile the project at the moment, so I can't get this added for a bit, but I'll track it here and try to slip it into a future release.

jakemastera, 2022-01-07:
> Upon investigating, as I expected the code to actually do something with the "Invert TFT" option checked is [only [...]

thanks for working on that so quickly! no worries, I ended up just flipping the housing that I mounted to the wall for now.

lbussy, 2022-01-07:
I may have left one out of the shipping boxes. I will look when I get home.

lbussy, 2022-01-08:
Well, as you likely suspected @thorrak, this is going to be a little more work than just adding that.  New libs, updated core, new fun!

thorrak, 2022-01-08:
> Well, as you likely suspected @thorrak, this is going to be a little more work than just adding that. New libs, updated core, new fun!

Is it? I'm surprised - I was actually expecting it to be ~a few lines (at least for the SSD1306 screens) but requiring a new semaphore to track status. You honestly would get 80% of the way there by just adding a check for invertTFT to [this line](https://github.com/thorrak/tiltbridge/blob/f0ed2cfd0ad45a1b3f4dcac845e7de1732db4564/src/bridge_lcd.cpp#L56)

lbussy, 2022-01-08:
I do not have the libraries downloaded that we can file with last time. New libraries will not compile cleanly, and I like the patience today to mess with it.

Guess which library threw the most errors?

thorrak, 2022-01-08:
LCBUrl?

lbussy, 2022-01-08:
No, our favorite one.

thorrak, 2022-01-11:
Well, I just got rid of that library in favor of one that you maintain, so... sorry @lbussy.

I was poking through a box earlier this afternoon and stumbled across an OLED board that I left here from last year. That said, I got this feature built, added, and tested on my end, so it should go in to the next release. 

@jakemastera - I'm working on getting a beta version out there as v1.0.3 shortly if you want to test it. Keep an eye on BrewFlasher/Fermentrack (or whatever you use to flash your firmware)

jakemastera, 2022-01-11:
> Well, I just got rid of that library in favor of one that you maintain, so... sorry @lbussy.
>
> I was poking through a box earlier this afternoon and stumbled across an OLED board that I left here [...]

Damn, you're fast! I'll keep an eye out for sure, thanks!

thorrak, 2022-01-11:
> Damn, you're fast! I'll keep an eye out for sure, thanks!

Hah, as are you! This should now be available as v1.0.3-Beta (for whichever flavor of board you have)

jakemastera, 2022-01-11:
> Hah, as are you! This should now be available as v1.0.3-Beta (for whichever flavor of board you have)

tested this morning and it worked perfectly, thanks for the quick response!

commit 45400346 (2022-01-11): Add "Invert Display" option for OLED (Closes #199)
closed 2022-01-13 by thorrak