thorrak/tiltbridge #206: MQTT only one of three active tilts show.

chrbratt, 2022-02-14:
Has a problem with only one of three active Tilt hydrometers being sent as a telegram to my mqtt broker from Tiltbridge. I'm a little unsure and do not really know in which end I should start troubleshooting. So I thought I would ask the question here and see that tiltbridge really has support for sending info from several tilts at the same time. ?

thanks in advance

[image]

[image]

chrbratt, 2022-02-14:
It send info for all three to Brewfather. 

[image]

madsmooth, 2022-02-16:
I only have a purple tilt and have not been able to get it to send any updates over mqtt. Any chance there isssie with some color codes preventing publish of updates?

thorrak, 2022-02-16:
> I only have a purple tilt and have not been able to get it to send any updates over mqtt. Any chance there isssie with some color codes preventing publish of updates?

That line of code I linked is the one that loops through all the colors, so it _shouldn’t_ matter — but it’s possible there is something else going on that causes it to somehow matter. 

If either of you have a spare ESP32 and don’t mind testing, there is a “tilt simulator” firmware I can push to BrewFlasher that will simulate all the colors of tilts simultaneously. Might be interesting to see if it’s the same colors dropping out

madsmooth, 2022-02-16:
> That line of code I linked is the one that loops through all the colors, so it _shouldn’t_ matter — but it’s possible there is something else going on that causes it to somehow matter.
>
> If either of [...]

I have a spare and willing to test. Is there a way to view log output using the web view?

thorrak, 2022-02-16:
Ah, hm. Unfortunately, unless you're comfortable compiling your own firmware it doesn't look like this will work, as it requires hardcoding your WiFi credentials. Let me see if I can get WiFiManager added in without too much effort.

thorrak, 2022-02-16:
I've thrown some code together here without any testing whatsoever (other than to make sure it compiles) so I can't promise that this works, but if you either launch BrewFlasher or use [BrewFlasher Web Edition](https://web.brewflasher.com/) you should see - under project "Other" board "ESP32" - an entry for "Tilt Sim v0.0.1". 

If it works, you'll see it spin up an AP named "TiltSimAP" with a password of "tiltsim". Log into it, connect it to your wifi, and it should begin to pretend to be _every_ tilt. 

If it doesn't work... say something here and I'll try to poke at the simulator code a bit tomorrow.

madsmooth, 2022-02-16:
I have a second set of the TFT hardware configuration. I was able to flash
the Tilt Sim via web flasher but never see an AP come up like you suggest.
Any chance you built it for a different hardware or perhaps it shouldn't
matter so long as it's esp32?

thorrak, 2022-02-16:
> I have a second set of the TFT hardware configuration. I was able to flash the Tilt Sim via web flasher but never see an AP come up like you suggest. Any chance you built it for a different hardware or perhaps it shouldn't matter so long as it's esp32?

It shouldn't matter, but I wasn't kidding when I mentioned my changes to add the wifi config support were **completely** untested. ;)

Let me test it later today and see if I can get that fixed

madsmooth, 2022-02-16:
No worries. I did end up testing with the v1.0.2 old firmware and mqtt was
sending updates for my purple tilt, when I updated back to v1.1 it did not
work for me, despite using the same settings. Still happy to test anything
in the future if you update the simulator.

thorrak, 2022-02-16:
The simulator actually should work as is. The AP may not have spun up if you used firmware that cached WiFi credentials in the past on that controller (e.g. TiltBridge) but it likely connected to WiFi and is sending out BLE iBeacon signals as if it is a bunch of Tilts. You can verify by launching the tilt app on your phone if you have it - the app will show them all.

I took a look at the changes to the MQTT code between 1.0.2 and 1.1.0 and unfortunately don't see anything jump out that would cause any errors. The MQTT keepalive setting was reduced (initially it was expressed in milliseconds, but the standard wants it expressed in seconds) -- but this shouldn't result in individual colors being skipped. Additionally, a "uniq_id" field was added to the Home Assistant autodiscovery MQTT messages -- but these aren't the messages you are (or should be) looking at/missing.

pletch, 2022-02-16:
Just a guess but is it possible that the MQTT buffer is now just a bit too small for the longer color names when attempting to publish the HA discovery payloads?  I _think_ overflowing this buffer will cause a MQTT client disconnect and the code only checks for connection on the first pass through the loop for the 3 topics for each color.  

The MQTT buffer size is set here:
[https://github.com/thorrak/tiltbridge/blob/v1.1.0/src/sendData.cpp#L10](https://github.com/thorrak/tiltbridge/blob/v1.1.0/src/sendData.cpp#L10)

madsmooth, 2022-02-16:
So now I understand what you mean. I tested with the old and recent
tiltbridge firmware and it was sending out 8 colors of mqtt messages
including homeassistant auto discovery messages and data messages.

I did notice the simulator has 16 tilts, 8 regular and 8 pro or HD in each
color but the bridge only identifies 8 colors, and it seemed like it was
sending a single message for each despite reporting from different
simulated devices. This may be how tiltbridge was designed so no issue from
me.

I turned off the simulator, rebooted tiltbridge and cleared mqtt broker
titlebridge message cache and tested again with nothing but my my real
purple tilt and now it works. Unfortunately I can't tell you why it didn't
work and now it does.

madsmooth, 2022-02-16:
You must be on to something. If I increase my mqtt topic more than 8
characters I get no mqtt traffic.

thorrak, 2022-02-17:
Changing this is trivial - I'll get a test version out later tonight. Thanks @pletch !

thorrak, 2022-02-18:
The test version is live. It's the v1.1.1 "beta" versions in BrewFlasher.

chrbratt, 2022-02-18:
It worked directly on the first test. :)

[image]

Many thanks!

thorrak, 2022-02-18:
Perfect! I'm going to reopen this for now until I can get this merged into master just in case anyone else has this issue before then, but hope to get this merged shortly. 

Thanks again @pletch for identifying the fix!

thorrak, 2022-02-18:
Unfortunately, I have run into that issue with another project, and haven't yet [found a solution](https://community.home-assistant.io/t/using-unique-ids-with-mqtt-autodiscovery-and-multiple-values-per-sensor/383269). If I find a solution (other than splitting the MQTT messages) I'll backport it to TiltBridge.

closed 2022-02-18 by thorrak

pletch, 2022-02-18:
@thorrak, I got curious about the autodiscovery issue and spun up a virtual machine to evaluate. Testing with mosquitto_pub / mosquitto_sub along with a home assistant core installation, I was able to get it to work.  I think the example [here](https://www.home-assistant.io/docs/mqtt/discovery/#sensors-with-multiple-values) is malformed and this is why it doesn't work.

If the discovery topics and messages are formatted as such, it will create both entities:
**Temperature Discovery Topic:**
_homeassistant/sensor/tiltbridge_tilt_Blue/temperature/config_
   **Message:**
{ 
"dev_cla": "temperature",
"unit_of_meas": "°C",
"ic": "mdi:thermometer",
"stat_t":"tiltbridge/tilt_Blue",
"name": "Tilt Temperature - Blue",
"val_tpl":"{{value_json.Temp}}",
"uniq_id": "tiltbridge_tiltBlueT" 
}

**Specific Gravity Discovery Topic:**
_homeassistant/sensor/tiltbridge_tilt_Blue/specgrav/config_
   **Message:**
{
"unit_of_meas": "SG",
"stat_t": "tiltbridge/tilt_Blue",
"name": "Tilt Specific Gravity - Blue",
"val_tpl": "{{value_json.SG}}",
"uniq_id": "tiltbridge_tiltBlueG"
}

Note that each entity must have its own uniq_id as well so I appended T or G accordingly.

There may be a way to get it to group the two entities under a single Tilt Blue MQTT discovered device but I am not sure how to do it at the moment.

[image]

chrbratt, 2022-02-18:
Wow, that did it @pletch!
@thorrak is it possible to get these changes in a test version.?

pletch, 2022-02-18:
@thorrak, I can set up a test install on a few ESP32s and push over a PR with some proposed updates if you don't have time to look at this.  LMK.

thorrak, 2022-02-19:
I just re-released v1.1.1 to incorporate this change as well. Thanks again for tracking this one down - I appreciate it!