thorrak/tiltbridge #210: Convert screen display to plato/brix

uSlackr, 2022-02-19:
Would you consider adding an option to display a calculated plato and/or brix value in lieu of SG?  I'm using Tb to push value to Home assistant and I am able to convert SG -> plato for display there.  It would be nice to be able to convert it for the LCD.

lbussy, 2022-02-19:
Speaking for me only ... maaaaybe?  There are quite a few places I can think of offhand where this would need to be addressed, and it begins to remove some of the simplicity (just use what you receive) of the TB functions.  If you *just* mean for the LCD display, that might be easier.

thorrak, 2026-03-18:
I just pushed a change that enables this. It will be released in the next v2.0.0 beta (likely in the next week or so).