thorrak/tiltbridge #212: Consider replacing AJAX polling on index and about pages with Websockets?

pletch, 2022-02-22:
@thorrak, curious if you have considered evaluating use of EspAsyncWebserver websockets push functionality to replace the [AJAX polling](https://stackoverflow.com/questions/10028770/in-what-situations-would-ajax-long-short-polling-be-preferred-over-html5-websock) done on the about page and index pages?  While playing with master to evaluate the MQTT fixes, I noticed there are still infrequent unrequested reboots that can occur when on the 'about' page.  

Over the weekend, I threw together a websockets implementation using the master branch and this qualitatively seems to work pretty well, perhaps being less resource intensive once the socket connection is established.  In addition, the values can be sent to multiple clients at once (though realize use case for this here is essentially non-existent) and the update frequency on the heap and uptime metrics can be increased to a few seconds with low overhead which is nice.

lbussy, 2022-02-22:
Those were my additions - and there were a few reasons I did not go with Websockets:
 1. The index page is a minor hit, Websockets would be unnecessary work.  A hammer when a scalpel is needed.
 2. The About page definitely creates resets on one of the models - but the fix is not avoiding AJAX but fixing whatever is crashing the device.  We left it there purposefully as a way to test the work we were doing to stabilize it.
 3. Nobody is going to sit there on the About page, the fun is on the Index.
 4. AJAX was a lot easier to develop on the client-side while I was working with it.
 5. With Async Web Server, there should be little difference in the overhead when we finally figure out what is causing the crashes (seems to be related to BT)

So, no reason not to do it, but there was just little reason to change for change's sake.

closed 2022-02-22 by pletch

thorrak, 2022-02-22:
@pletch Do you have an example using websockets? I'd be curious to see it.

In an alternate project I've been playing with using the JSON-specific AsyncWebServer endpoints, and am contemplating porting that functionality over to TiltBridge.

pletch, 2022-02-22:
I only have the very crude proof of concept branch that I put together to play with websockets implementation for the about page.  I am happy to push up and share if interested.  If you want a proper working example, I don't have anything to share at the moment.  

For educational purposes, I will eventually work on more robust websocket implementation for the dual esp32 Tiltbridge design I put together last year while backporting some of the other recent changes introduced here in v1.1.1

thorrak, 2022-02-22:
> I only have the very crude proof of concept branch that I put together to play with websockets implementation for the about page. I am happy to push up and share if interested. If you want a proper [...]

If you don't mind pushing it to GitHub, I would absolutely be interested in taking a look and seeing if there is anything that could be adopted. Thanks!

pletch, 2022-02-22:
Done!  It is a branch called websockets under my fork.