thorrak/tiltbridge #213: [Feature Request] Support ESPHome.io Flashing

conallob, 2022-03-15:
Create an esphome.io compatible yaml config, to support flashing via ESPHome (and by extension home-assistant.io managed ESP devices)

thorrak, 2022-03-15:
Do you have a link to documentation on this? I've never heard of this before.

conallob, 2022-03-15:
Of course

* http://www.esphome.io is the main site
* https://esphome.io/guides/getting_started_hassio.html is the guide for using ESPHome via home-assistant.io
* https://github.com/klaasnicolaas/home-assistant-glow is an example of an ESP32 based project on GitHub with easy integration into home-assistant.io

thorrak, 2022-03-15:
Unfortunately, from reading the documentation and looking at the home assistant glow project, I don't think that ESPHome and TiltBridge are compatible projects. The YAML you are referencing seems to be in order to configure basic GPIO functions on individual ESP pins, and TiltBridge requires a substantial amount of custom code in order to function.

Sorry about that. :(

closed 2022-03-15 by thorrak

conallob, 2022-03-16:
Perhaps https://github.com/klaasnicolaas/home-assistant-glow was too complex an example, it was just the other ESP32 project I know of with good Github integration.

https://www.reddit.com/r/homeassistant/comments/exmq1p/tilt_hydrometer_through_esphome/ shows it is possible to use ESPHome.io to manage a ESP32 reading a Tilt hydrometer.

I'm happy to take this issue and create a pull request with a yaml config

lbussy, 2022-03-16:
I also think there's a potential here, John.  Hast will consume data as well as control, so a read-only configuration should be possible.  For those using Hast, this would be a real nice integration.

thorrak, 2022-03-16:
That’s interesting — in my head, based on the home assistant glow example it felt like ESPHome was taking a yaml file to configure an existing binary to read/output/interact with data available on a pin. From the linked Tilt example, it sounds like you can actually get new C/C++ code running using this method, which would imply that TiltBridge could be reformatted to support this. 

I’m still a bit confused/skeptical as to what you gain from this, however. TiltBridge already supports MQTT (and HASS by extension). Flashing is already supported by BrewFlasher and BrewFlasher Web. If the *code* needs to be rewritten/restructured in any way to support this - which the Tilt Hydrometer integration linked makes it seem like it would - then the project ends up becoming permanently targeted at ESPHome, which I’m not a fan of. 

If all that is required here is adding a YAML file (which is ignored by PIO) then I’ve got no objections to this if you want to raise a PR — I’m all for increasing compatibility with as many things as we can. While it may just be my lack of imagination though, that doesn’t feel like all this would require.

conallob, 2022-03-16:
ESPHome integration within Home Assistant gives you a unified UI for managing ESP32 devices. It would be an alternative to BrewFlasher, which automatically flags when there is an update to be pushed to each ESP32.

I would agree that refactoring the code to support this feature is overkill. I'm happy to scope this issue to just creating a yaml file.

thorrak, 2022-03-16:
Yeah, this sounds like a separate project, even if the functionality is closely related. It sounds like it isn't really needed given how you describe ESPHome, but I'd be happy to add support for flashing this firmware (if necessary) to BrewFlasher.