thorrak/tiltbridge #216: Recent compilation failing

mianos, 2022-07-04:
I just installed platform io from scratch and I get the following error:
```

Processing lcd_ssd1306 (board: lolin_d32; platform: espressif32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------
Configuring toolchain packages from a remote source...
Error! Failed to extract upstream toolchainconfigurations:
Bad package version `1.22.0-97-gc752ad5-5.2.0`
You can disable this feature via the `board_build.arduino.upstream_packages = no` setting in your `platformio.ini` file.
The terminal process "C:\Users\hotma\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
``
I am not 100% sure but it seems that there is no matching extensia compilers for the hard coded Espressive #1.0.6

Has anyone else had this issue? Is there a simple fix?

thorrak, 2022-07-04:
Weird.

I haven't tested it on an actual TiltBridge, but I think I just made the necessary changes to the [devel branch](https://github.com/thorrak/tiltbridge/tree/devel) to enable Arduino 2.x support and remove the locked platform package requirement. Pull that branch and give it a shot.

mianos, 2022-07-04:
All the versions build fine now. Thanks. I'll have to dig up a board to try it out.

closed 2022-07-26 by thorrak