thorrak/tiltbridge #218: 1.1.3-beta webgui "too many redirects" error

bsaurusrex, 2022-08-09:
D32-pro tft.  TFT turns on ok, connects to wifi but the webgui shows too many redirects error
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None

thorrak, 2022-08-09:
This sounds like the SPIFFS partition didn't flash correctly. I just reflashed a device with the latest firmware (updated as of ~2 hours ago) and the latest BrewFlasher (1.4, though I saw the other issue you raised) and it seems to be working on my end. If you can, I'd recommend reflashing again.

bsaurusrex, 2022-08-09:
Since v1.0.1, I havent been able to successfully flash.  This latest v1.1.3-beta, I finally was able to flash it and get the tft to display.

I ended up getting brewflasher 1.0 to open then I selected wipe data this time and was able to setup from scratch.  Unfortunately that led to constant crashes and resets (tiltbridgeAP mode)

Trying to downgrade to 1.1.2 with data wipe now.

bsaurusrex, 2022-08-09:
1.1.2 with wipe all data, no tft and no tiltbridgeAP.

Is it possible I have an hardware issue? I bought from the aliexpress lolin store.

thorrak, 2022-08-09:
While it's possible, I'd say that it's unlikely. There have been two bugs that have created issues with the D32 pro version of TiltBridge -- one with TiltBridge itself, and one with BrewFlasher.

BrewFlasher 1.4 was released specifically to fix the BrewFlasher issue. This rolled back the esptool version that was causing the D32 Pro firmware to fail to flash to 16MB devices, and - in testing - works on my end with the v1.1.3-beta firmware.

TiltBridge v1.1.3-beta (the version I released a few hours ago) fixes the issue with WiFiManager that was causing periodic boot loops (the "white screen" issue). I've tested that on my end, and it seems to be working perfectly. Version 1.1.2 almost certainly _does not_ work, so I'm not surprised you aren't seeing the AP spin up after flashing.

bsaurusrex, 2022-08-09:
Seems to be working now with v1.1.3-beta.  All others do not boot except 1.0.1.

I did notice another bug though, when I put mqtt as a dns and not an IP it goes into a boot loop and I have to reflash it.  Otherwise all looks to be working now.

closed 2022-08-09 by bsaurusrex