thorrak/tiltbridge #220: Request: New Hardware Support

SillyHats, 2022-08-25:
A faster chip with 16MB of flash and bigger, higher pixel density screen version of the TTGO USBC board is now available.

https://github.com/Xinyuan-LilyGO/T-Display-S3

https://www.aliexpress.com/item/3256804310228562.html

https://www.youtube.com/watch?v=j8THAc1sMww

thorrak, 2022-08-25:
Thanks for sending this across! This is a really need looking board, and I can see it being incredibly useful in a number of projects.  Unfortunately, I'm unlikely to build/maintain support for this board in TiltBridge - there are two main reasons why:

First, it's based on the ESP32-S3. I've ranted about this to anyone that will listen, but Espressif unfortunately has named all of their latest chips "ESP32"something, even when they are based on completely different architectures. Although I'm not averse to supporting chips other than the base ESP32, library support isn't guaranteed, so this ends up being quite a bit of work (and potentially a fair amount of refactoring as well).

Second, the screen resolution is 170x320, which means that the user interface would need to be redesigned from the one in use for the LoLin TFTs (and all the generic versions thereof). This is actually more of a pain than trying to target a new chipset, as while suppoting a new chipset generally means doing a single pass through old code, supporting a new screen resolution means every _future_ change to TiltBridge would require one additional set of revisions. 

That said, if they end up releasing a version with a 240x320 screen, let me know! Given the historical supply constraints on LoLin TFTs, I'm happy to try to support something new if it seems sustainable!

closed 2022-08-25 by thorrak

SillyHats, 2022-08-25:
Thanks for the detailed reply!