thorrak/tiltbridge #221: Bug with i2c_device_at_address method.- new Heltec board has no OLED output

mmissey, 2022-12-05:
Hey @thorrak, really appreciate you building this firmware. Super cool!

Just wanted you and anyone else that might end up here about an issue with the HTIT-WB32 board I purchased [here](https://www.amazon.com/dp/B076KJZ5QM?psc=1&ref=ppx_yo2ov_dt_b_product_details

Installing any of the firmwares via Brewflasher resulted in a completely blank OLED screen. To resolve this I cloned the repo and edited the lcd code [here](https://github.com/thorrak/tiltbridge/blob/0152d603fa9f061aa02eb3dec8aa779d42ec8dc2/src/bridge_lcd.cpp#L45) to remove the `if/else if`  and replaced with just the TTGO style block. Now it works after compiling and uploading with pio.

I did some debugging and it seems like the method `i2c_device_at_address` is blocking any subsequent calls if it throws an error on any previous calls.

I'd attempt a fix and submit a PR, but I don't have a great understanding of how these Arduino libraries are working and I only have the 1 device to test on.

thorrak, 2022-12-06:
My guess is that the issue is somewhere [in here](https://github.com/thorrak/tiltbridge/blob/0152d603fa9f061aa02eb3dec8aa779d42ec8dc2/src/bridge_lcd.cpp#L536-L538), which is to say that it's almost certainly in the way that the Wire library is interacting with the (incorrect) pins.

Was the only issue that the screen didn't work, or was the issue that the entire board failed to boot/detect anything?

thehijjt, 2023-01-05:
I am having this issue with the latest heltec board that I purchased it is labelled WiFi_Kit_32_V3 [https://www.amazon.com/dp/B07DKD79Y9?psc=1&ref=ppx_yo2ov_dt_b_product_details](url). It looks identical to the link posted, I am not sure how mmissey got this to work without doing a lot more than just changing the bridge_lcd. I made those changes as well as the changes below to get the firmware loaded. I still get a blank screen, but the TiltbridgeAP shows up and allows connection. 

1. Make sure that espressif32 platform was updated, and platformio was the latest 
2. I had to change the build configuration to `board =esp32-s3-devkitc-1` 
3. Upgrade NimBLE to a version that supports the esp32-s3 I just went to the git version, I don't know if this works 

I know that initially the screen was working, so I gave it a go with some generic code for scanning the i2c bus. Scanning didn't seem to be working with the default libraries, I was never seeing a 0x3c at 4, 15 which from the images on the listing should be the correct pins. 

Brewflasher doesn't work on this at all because the chip is esp32-s3, did this originally come with a different chip?

mmissey, 2023-01-05:
Hey @thehijjt,

From the debugging I was doing, using the default `Wire` library to scan for wires that accept a transmission doesn't work. That library seems to lock out on failure. ie: You scan one config and it fails and then no transmissions after that work.  
Even without my changes to the bridge_lcd init method the firmware still loaded, but with no screen.
 
I'd suggest you look up the spec sheet for the board you got but I _think_ it's got the same pinout as mine so it should work if that init function has 
```c
// For the "TTGO" style OLED shields, you have to power a pin to run the backlight.
pinMode(16, OUTPUT);
digitalWrite(16, LOW); // Set GPIO16 low to reset OLED
delay(50);
digitalWrite(16, HIGH); // While OLED is running, must set GPIO16 in high
oled_display = new SSD1306Wire(0x3c, 4, 15);
```

I'll double check which build config I used and edit this comment

thehijjt, 2023-01-05:
I got it to work, I understood what you did, but I thought you had the same hardware, however I believe we have entirely different hardware. I got the display to work using the following i/o
`        pinMode(21, OUTPUT);
        digitalWrite(21, LOW); // Set GPIO16 low to reset OLED
        delay(50);
        digitalWrite(21, HIGH); // While OLED is running, must set GPIO16 in high
        oled_display = new SSD1306Wire(0x3c, 17, 18);`

I also removed the i2c_device_at_address call, I am pretty sure it was boot looping because of that... 

This is the schematic was available on the heltec website, the picture in the amazon listing was very incorrect. 
[diagram of the v3](https://resource.heltec.cn/download/WiFi_Kit_32_V3/HTIT-WB32_V3_Schematic_Diagram.pdf)

Thanks for the advice, I am not sure if there are other issues, but at least it is displaying now.

thorrak, 2023-01-05:
Thanks for this — I’m working on a refactor of the display code this week and hope to have something out this weekend.

thorrak, 2023-01-06:
> I got it to work, I understood what you did, but I thought you had the same hardware, however I believe we have entirely different hardware. I got the display to work using the following i/o [...]

Let me take a closer look at the schematic, but at first glance, was the write to pin 21 required?

thehijjt, 2023-01-06:
It was required to get the display to turn on. It just remains dark without turning that on. I originally had a different gpio and changing it to that on a whim caused the display to work. Although, when I turn it on now sometimes the display is scrambled and won't display anything unless I hit the reset button, then it works fine.

closed 2023-01-09 by thorrak

uSlackr, 2024-04-01:
Is there a version of tiltbridge firmware that works with the Heltec WiFi_Kit_32_V3 now?  I'm trying to flash OLED 1.2.2 with brewflasher and it fails to install  with  "esptool raised an error"

uSlackr, 2024-06-22:
Based on other threads, I now know the issue with v3.