thorrak/tiltbridge #222: ESP32-D0WD-V3 (revision v3.0)

MaligatorX, 2022-12-18:
Hi may i know is it possible to install firmware on "ESP32-D0WD-V3 (revision v3.0)" board?
is there any one to help me? 
Regards,

thorrak, 2022-12-18:
Unfortunately, the OLED screens are separate from the ESP32 chips. There is no reason I know of that a ESP32-D0WD-V3 shouldn’t work — but that doesn’t mean that the OLED screen your board came with is compatible.

MaligatorX, 2022-12-18:
Thanks for your early reply, OLED is not a important i can disconnect it, i have no ssid in wifi network

thorrak, 2022-12-18:
Can you link the exact board you purchased?

MaligatorX, 2022-12-19:
i got this board [here](https://eshop.eca.ir/%D9%85%D8%A7%DA%98%D9%88%D9%84-%D8%B4%D8%A8%DA%A9%D9%87-%D9%88-wifi/7255-%D8%A8%D8%B1%D8%AF-%D8%AA%D9%88%D8%B3%D8%B9%D9%87-esp32-%D8%AF%D8%A7%D8%B1%D8%A7%DB%8C-wifi-%D9%88-%D8%A8%D9%84%D9%88%D8%AA%D9%88%D8%AB-30-%D9%BE%D8%A7%DB%8C%D9%87.html) 

**Introduction**

The ESP-WROOM-32 is a universal Wi-Fi+BT+BLE MCU module that is powerful and versatile for low-power sensor networks and demanding tasks such as voice coding, audio streaming and MP3. Decoding and so on. The core of this module is the ESP32-D0WDQ6 chip*, which is scalable and adaptive. Two CPU cores can be individually controlled or powered up. The clock frequency can be adjusted from 80 MHz to 240 MHz. The user can power off the CPU and use the low-power coprocessor to continuously monitor the state change of the peripheral or whether certain analogs exceed the threshold. The ESP32 also integrates a wide range of peripherals including capacitive touch sensors, Hall sensors, low noise sense amplifiers, SD card interfaces, Ethernet interfaces, high speed SDIO / SPI, UART, I2S and I2C.

**Specification**

The ESP32-D0WDQ6 has two low-power Xtensa® 32-bit LX6 MCUs. On-chip storage includes:
• 448 KB of ROM for program startup and kernel function calls.
• 520 KB of on-chip SRAM for data and instruction storage.
• 8 KB of SRAM in the RTC, RTC slow memory, which can be accessed by the coprocessor in Deep-sleep mode.
• 8 KB of SRAM in the RTC, RTC Express Memory, which can be used for data storage and access by the host CPU during RTC startup in Deep-sleep mode.
• 1 kbit eFuse, of which 256 bits are system specific (MAC address and chip settings); the remaining 768 bits are reserved for user applications, including Flash encryption and chip ID.
• ESP-WROOM-32 integrates 4MB of SPIFlash

thorrak, 2022-12-19:
That controller should work fine. Interestingly enough, that’s one of the controllers I originally used to design the first version of TiltBridge. 

Which versions of the firmware have you tried so far? Have you tried V1.1.1 OLED?

MaligatorX, 2022-12-19:
Thanks for Your reply's,

I flashed V1.1.1 OLED and the display shows OK but still has no WIFI SSID so i get another ESP32 from that web site and flash with 1.1.1 then i could see SSID in my network, it seems first bords WIFI module was corrupted, but when i flash vv1.1.3 i has no OLED but wifi works fine 
if you can make a change in firmware that OLED works with 1.1.3 it will be perfect .

Thanks again for your help,

thorrak, 2022-12-19:
I apologize as I think I may have misunderstood.

Do you have a separate OLED module? If so, which module? How do you have it connected? Do you know the module’s I2C address?

MaligatorX, 2022-12-19:
Yes its a SSD1306 0.96 inch OLED 
You can find more technical info here[here](https://server4.eca.ir/eshop/ECA_Modules/index.php?d=OLED-I2C-SSD1306%5B3011002008%5D&s=n&o=asc) 

Thanks a lot

thorrak, 2022-12-19:
Can you confirm what pins you have wired to which inputs on the screen?

MaligatorX, 2022-12-19:
Sure, its d21 and d22 for  sda/slc and other two for power 3v3/gnd

thorrak, 2023-01-09:
I believe this will be fixed when #224 gets merged (hopefully in the next 24 hours)

closed 2023-01-09 by thorrak