thorrak/tiltbridge #226: add support for Lilygo TTGO S3

robertvandervoort, 2023-03-24:
It would be a great upgrade from the T-display which is supported to this one. The ESP32 S3 is supported in the later version of esptool which I don't believe brewflasher is using yet.

thorrak, 2023-03-24:
I’m not really looking to support more hardware at this point, but I’m also not opposed to supporting it unofficially. Do you have a link to the specific board you are interested in?

SillyHats, 2023-03-27:
I'm not @robertvandervoort but I think this is the board in question https://www.aliexpress.us/item/3256804310228562.html

thorrak, 2023-03-27:
Looks interesting -- I just ordered one. No promises, but we'll see!

thorrak, 2023-06-04:
Just as an update to this - I received my board a bit over a month ago, but haven’t had a chance to play with it just yet. First impressions: Physically, it seems like a neat design factor!

uSlackr, 2024-05-30:
Any update on this?  The heltec wifi kit 32 v3 uses the S3 as well.

robertvandervoort, 2024-09-09:
My apologies for not seeing that you replied to my ask. Thank you for considering it! The other one's display was too hard for me to read though the code works fantastic and I've been using it to track more than a dozen brews since March of last year. Appreciate you for doing what you're doing!