thorrak/tiltbridge #240: Does an ideaspark® ESP32 OLED also fit?

fkroepfl, 2024-01-05:
Does an ideaspark® ESP32 Development Board with 0.96 Inch OLED Display,CH340,WiFi+BLE also fit to install Tiltbridge?

e.g. https://www.ebay.com/itm/176053628906

thx.

/FJ

[image]

thorrak, 2024-01-05:
Although I have never used this board and thus cannot guarantee anything, it does appear that I already attempt to initialize the correct pins this board requires: https://github.com/thorrak/tiltbridge/blob/e06746762e5ab6f6f713ceaebae0e58a7bda94e4/src/bridge_lcd.cpp#L72

fkroepfl, 2024-01-05:
That sounds good ... 🤗
Thank you very much for the code x-check.

closed 2024-01-05 by fkroepfl