thorrak/tiltbridge #241: Tilt data is rarely to never sent to Grainfather Platform

fkroepfl, 2024-01-29:
The logging URLs for 2 tilts (red, blue) are entered on the Grainfather platform.  

Only a single data point has been transmitted for one tilt so far, and no data point at all for the other tilt.

Locally, the tilt data is visible via BLE and via MQTT in Home Assistant.

fkroepfl, 2024-01-31:
As already mentioned, I am trying to operate two tilt devices, RED and BLUE, via TiltBridge.

Tilt BLUE has not been sending data to Grainfather via logging URL for nine hours, or irregularly, i.e. not at the specified interval, via MQTT to Home Assistant. 

However, and this makes me wonder, Tilt BLUE sends data via Bluetooth at very short intervals, which can be received locally via other devices, eg Home Assistant via BT Interface.

In the meantime, I'm pretty stumped, although it now looks like TiltBridge is having problems proxying the data.

Or am I seeing this wrong?

thorrak, 2024-02-01:
Is the logging URL you are using HTTP or HTTPS?

thorrak, 2024-02-01:
Unfortunately, Grainfather support is something that someone contributed to TiltBridge, and I don't have any personal experience with it or with the Grainfather platform. Reading through the code I don't see anything obvious that looks like it should be a problem, other than potentially interactions with other send targets. I have a few questions for you in order to help me try to diagnose what is going on:

1. What other send targets are you using, if any? You mention MQTT - what is the send delay for that -- and are you using TiltBridge for MQTT?
2. You mention something about TiltPi -- Are you currently running both TiltPi and TiltBridge? 
3. Are you using a single TiltBridge for both the Blue and Red tilt, or multiple?
4. When you look at the "about" page for TiltBridge, what is the uptime that you see? Is the TiltBridge crashing?

fkroepfl, 2024-02-01:
FYI: This restart was definitely not triggered by me.

> Uptime:
> Days: 0, Hours: 3, Minutes: 4, Seconds: 55
> Reset Reason:
> Reason: ESP_RST_SW, Description: Software reset via esp_restart
> Heap Information:
> Free Heap: 72140, Max: 36852, Frags: 49

thorrak, 2024-02-02:
Yeah -- that one (with the reason "Software reset via esp_restart") is a timed reset that was added awhile back when there were issues with one of the libraries which could cause the device to progressively slow down. I've since removed that library, but admittedly forgot that reset was in there. Still though - it doesn't hurt to have it. ;)

Unfortunately, at the moment I'm at a loss. At the moment, the connections don't have any kind of logging attached to them to show the status. This is something that would be a good idea, though, so I'll try to get it added with the upcoming TiltBridge refresh. I'm tracking that feature request [here](https://github.com/thorrak/tiltbridge/issues/242).

We're travelling for the next few days, so I'm not sure if I'll be able to get something built to try to debug what is happening here, but in the mean time my recommendation would be to potentially temporarily disable the MQTT send on TiltBridge and see if that helps. It's possible that there is something weird going on with that target which is causing Grainfather not to work.

fkroepfl, 2024-02-02:
> Unfortunately, at the moment I'm at a loss

I feel the same way 😉 .

I'm still trying to find out more by replacing components. I now have a new ESP32 board, without OLED display but with the same firmware, in use and only started with Tilt-BLUE. Interestingly, it has worked with Grainfather so far. After about 4 hours I also activated Tilt-RED. This has also worked in principle so far. Only one data point was lost - for whatever reason.

However, it is noticeable that the data series via MQTT are now very poor. A lot of values are missing.

What is not clear to me is whether it is the ESP32 module or the combination of data targets that influences or triggers the behavior.

thorrak, 2024-02-02:
Huh. Out of curiousity, if you set the Red tilt to sleep mode (rest it vertically) does the Blue tilt update properly?

I'm wondering if something may be going on with the rate limiting on Grainfather's side. If that fixes it,  I can reach out to them to find out.

fkroepfl, 2024-02-03:
> Huh. Out of curiousity, if you set the Red tilt to sleep mode (rest it vertically) does the Blue tilt update properly?

No, but if you start with blue alone after a reset, the values are identical.

If red then also transmits, the data is swapped.

[image]

[image]

fkroepfl, 2024-02-03:
@thorrak A few more comments.

If the Tilt Red is active first, or the Red and Blue are operating at the same time, or the Tilt Blue is not transmitting alone and before the Tilt Red, the data sent via TiltBridge will not be interpreted correctly.

Either they are not updated or, as already documented, they are mixed up when Blue is activated first.

[image]

thorrak, 2024-02-04:
Hmm. I'm at a bit of a loss as to what could be happening here. The code looks correct, but clearly the outcome you're seeing is wrong which indicates that something obviously isn't working. 

After reading through everything, even ignoring this issue I think there is an opportunity to clean things up. I went ahead and rebuilt this morning, and am going to spend this afternoon testing them to make sure they work (and don't leak memory all over the place). I'll try to get a "test" firmware built for you to try out later tonight assuming all goes well.

With that in mind, which firmware type do you use? OLED, TFT, or eSPI?

fkroepfl, 2024-02-04:
🤗
> With that in mind, which firmware type do you use? OLED, TFT, or eSPI?

It's OLED Firmware on two different ESP Boards, one with display, one without.

thorrak, 2024-02-04:
Alrighty! The new beta firmware should now be available on BrewFlasher with the rewritten data sender: https://web.brewflasher.com/fw/25

The version is TiltBridge Beta - v1.2.2-beta1 - OLED

I've tested this with a single Tilt with Grainfather and can see that it updates, but unfortunately don't have more colors here at the moment to test with. 

As always, please let me know if you have any issues/questions (or alternatively, if this solves the issue). Once you confirm that things are good (fingers crossed!) then I will merge the code into master and issue an official release.

fkroepfl, 2024-02-06:
@thorrak I have now installed the FW and will share my observations after a while.

It should be mentioned briefly that with the current FW version you have to wait a disproportionately long time for a response when calling up the settings page.  

http://192.168.1.192/settings/

thorrak, 2024-02-06:
Unfortunately, I can't replicate the slow response time on my end. When I go to the settings page it fully loads within 2-3 seconds. It might be worth restarting the device at some point and seeing if this helps.

fkroepfl, 2024-02-06:
> Unfortunately, I can't replicate the slow response time on my end. When I go to the settings page it fully loads within 2-3 seconds. It might be worth restarting the device at some point and seeing if this helps.

It was noticeable after the first boot after flashing.
After a power reset, this behavior was no longer present.

fkroepfl, 2024-02-06:
It looks good for the first monitoring use. The values that come from the same tilt and are only forwarded via different paths are now identical.

Only Grainfather is now set up with 2 uncalibrated tilts.

The next step is to setup MQTT as target and finally the calibration.

If I notice anything else during operation, I'll let you know.

Many thanks for the bugfix. 🤗

**Explanation of the labels:**

- Blue Tilt => Tilt Blue@TiltPi
- Green Tilt => Tilt Blue@TiltBridge
- Red Tilt => Tilt Red@TiltPi
- Yellow Tilt => Tilt Red@TiltBridge

[image]

fkroepfl, 2024-02-06:
@thorrak Since there have been no significant problems with reception on Grainfather with the last FW version, I will create a separate post for the MQTT problems.

thorrak, 2024-02-06:
Glad to hear that this solves the Grainfather sending issues. I'll tag this issue to close when that is merged.

> Only Grainfather is now set up with 2 uncalibrated tilts.

What do you mean by this comment? Do you mean that you do not currently have the tilts calibrated within TiltBridge, or that the calibrated data isn't being used (only the raw data is being sent to Grainfather)?

fkroepfl, 2024-02-06:
FYI: The data series in direct comparison

[image]

fkroepfl, 2024-02-06:
> What do you mean by this comment? Do you mean that you do not currently have the tilts calibrated within TiltBridge, or that the calibrated data isn't being used (only the raw data is being sent to Grainfather)?

I meant to say that I don't use the calibration feature in either TiltBridge or TiltPi so that I have the raw data consistently across both ways for comparison.

commit 0d23e3c6 (2024-02-04): Refactor Grainfather support to use send_to_url
(Fixes #241)
commit e4aac910 (2024-02-04): Refactor Grainfather support to use send_to_url
(Fixes #241)
closed 2024-03-03 by thorrak