thorrak/tiltbridge #248: Adding additional information to the Home Assistant MQTT Auto Discovery messages

fkroepfl, 2024-02-15:
The aim is to optimize the visualization of tilt devices and sensor data in the Home Assistant by modifying the auto discovery messages.

### **Initial situation using the "Tilt Blue" example:**

```
homeassistant/sensor/tiltbridge_tilt_Blue/temperature/config

{
  "device": {
    "identifiers": "Blue",
    "name": "Tilt Blue"
  },
  "dev_cla": "temperature",
  "unit_of_meas": "°C",
  "ic": "mdi:thermometer",
  "stat_t": "tiltbridge/tilt_Blue",
  "name": "Tilt Temperature - Blue",
  "val_tpl": "{{value_json.Temp}}",
  "uniq_id": "tiltbridge_tiltBlueT"
}

```
### **Suggestion:**

**1) Add an attribute to the sensor status as secondary information, e.g.: the timestamp**

References found to: `state.last_updated` & `state.last_changed`

- https://www.home-assistant.io/docs/configuration/state_object/
- https://www.home-assistant.io/dashboards/entities/#secondary_info

If the sensor now has "temperature" as the status and, for example, the timestamp as an attribute, an update with the same temperature but a different timestamp will update last_updated but not last_changed, as only the attribute has been updated.

The same would of course also apply to the "Specific density".

_Here is an example of the effect in the entity card_

[image]

**2) Additional information about the device to which this sensor belongs**

References found for configuration variable: device

- https://www.home-assistant.io/integrations/sensor.mqtt/#device

Could be of interest IMHO:
- https://www.home-assistant.io/integrations/sensor.mqtt/#configuration_url
- https://www.home-assistant.io/integrations/sensor.mqtt/#manufacturer
- https://www.home-assistant.io/integrations/sensor.mqtt/#model
- https://www.home-assistant.io/integrations/sensor.mqtt/#via_device

thorrak, 2024-02-16:
As I mentioned, I don't use either MQTT or Home Assistant -- but at one point I set up a HAST instance with the intent of trying to get it working with TiltBridge. Unfortunately, Home Assistant Autodiscovery wasn't (isn't?) the best documented at the time, and so I ended up giving up.

While I can't promise to get this one fixed soon, I'll give it a shot when I have some time.

fkroepfl, 2024-02-17:
FYI: With the following manual configuration (YAML/JSON) it was possible to choose the display of the sensor status according to the HA options (screenshot).

I hope it helps when thinking about how to integrate this into the code.

[image]

[image]

**mqtt.yaml**
```
sensor:
# Red Tilt
# Temperature
  - name: "MyTilt|Red Temperature"
    unique_id: "myTiltRedTemperature"
    unit_of_measurement: "°C"
    icon: "mdi:thermometer-water"
    state_topic: "tiltbridge/tilt_Red"
    value_template: "{{value_json.Temp}}"
    json_attributes_topic: "tiltbridge/tilt_Red"
    json_attributes_template: >
      { "Uptime": "{{ value_json.timeStamp }}" }
    device:
      name: "Tilt Red"
      identifiers: "Red"
      model: "TILT™ HYDROMETER AND THERMOMETER"
      manufacturer: "Baron Brew Equipment LLC"
      configuration_url: "http://tiltbridge.local/"
      via_device: "TiltBridge"
      sw_version: "v1.2.1"
      suggested_area: "Brewery"

# Specific Gravity
  - name: "MyTilt|Red Specific Gravity"
    unique_id: "myTiltRedSpecificGravity"
    unit_of_measurement: "SG"
    icon: "mdi:slope-downhill"
    state_topic: "tiltbridge/tilt_Red"
    value_template: "{{value_json.SG}}"
    json_attributes_topic: "tiltbridge/tilt_Red"
    json_attributes_template: >
      { "Uptime": "{{ value_json.timeStamp }}" }
    device:
      name: "Tilt Red"
      identifiers: "Red"
      model: "TILT™ HYDROMETER AND THERMOMETER"
      manufacturer: "Baron Brew Equipment LLC"
      configuration_url: "http://tiltbridge.local/"
      via_device: "TiltBridge"
      sw_version: "v1.2.1"
      suggested_area: "Brewery"

# Uptime
  - name: "MyTilt|Red Uptime"
    unique_id: "myTiltRedUptime"
    unit_of_measurement: "s"
    icon: "mdi:timer-sync"
    state_topic: "tiltbridge/tilt_Red"
    value_template: "{{ value_json.timeStamp }}"
    device:
      name: "Tilt Red"
      identifiers: "Red"
      model: "TILT™ HYDROMETER AND THERMOMETER"
      manufacturer: "Baron Brew Equipment LLC"
      configuration_url: "http://tiltbridge.local/"
      via_device: "TiltBridge"
      sw_version: "v1.2.1"
      suggested_area: "Brewery"

```

**JSON (yaml converted)**
```
[
  {
    "name": "MyTilt|Red Temperature",
    "unique_id": "myTiltRedTemperature",
    "unit_of_measurement": "°C",
    "icon": "mdi:thermometer-water",
    "state_topic": "tiltbridge/tilt_Red",
    "value_template": "{{value_json.Temp}}",
    "json_attributes_topic": "tiltbridge/tilt_Red",
    "json_attributes_template": "{ \"Uptime\": \"{{ value_json.timeStamp }}\" }\n",
    "device": {
      "name": "Tilt Red",
      "identifiers": "Red",
      "model": "TILT™ HYDROMETER AND THERMOMETER",
      "manufacturer": "Baron Brew Equipment LLC",
      "configuration_url": "http://tiltbridge.local/",
      "via_device": "TiltBridge",
      "sw_version": "v1.2.1",
      "suggested_area": "Brewery"
    }
  },
  {
    "name": "MyTilt|Red Specific Gravity",
    "unique_id": "myTiltRedSpecificGravity",
    "unit_of_measurement": "SG",
    "icon": "mdi:slope-downhill",
    "state_topic": "tiltbridge/tilt_Red",
    "value_template": "{{value_json.SG}}",
    "json_attributes_topic": "tiltbridge/tilt_Red",
    "json_attributes_template": "{ \"Uptime\": \"{{ value_json.timeStamp }}\" }\n",
    "device": {
      "name": "Tilt Red",
      "identifiers": "Red",
      "model": "TILT™ HYDROMETER AND THERMOMETER",
      "manufacturer": "Baron Brew Equipment LLC",
      "configuration_url": "http://tiltbridge.local/",
      "via_device": "TiltBridge",
      "sw_version": "v1.2.1",
      "suggested_area": "Brewery"
    }
  },
  {
    "name": "MyTilt|Red Uptime",
    "unique_id": "myTiltRedUptime",
    "unit_of_measurement": "s",
    "icon": "mdi:timer-sync",
    "state_topic": "tiltbridge/tilt_Red",
    "value_template": "{{ value_json.timeStamp }}",
    "device": {
      "name": "Tilt Red",
      "identifiers": "Red",
      "model": "TILT™ HYDROMETER AND THERMOMETER",
      "manufacturer": "Baron Brew Equipment LLC",
      "configuration_url": "http://tiltbridge.local/",
      "via_device": "TiltBridge",
      "sw_version": "v1.2.1",
      "suggested_area": "Brewery"
    }
  }
]
```

thorrak, 2024-02-18:
The YAML here is incredibly helpful, as I think I understand how it maps to the sensor configuration payloads that TiltBridge is sending (and where the differences are) which is a big part of the trouble I was facing. In the device announcements we're [abbreviating](https://github.com/home-assistant/core/blob/dev/homeassistant/components/mqtt/abbreviations.py) most of the keys, but mapping them shouldn't be an issue.

With the changes that I made for  #247 I think that adding (at least some of) the `device` keys should be pretty straightforward. I'm going to try doing it in a way that is at least _theoretically_ backwards compatible so as to not have to wait for a major version, but if this doesn't seem to work let me know and I can tweak further.

fkroepfl, 2024-02-19:
>  Most of the buttons are abbreviated in the device announcements, but the assignment should not be a problem.

Since I'm not familiar with the abbreviation rules, I operated with the full names 😉

>  I'm going to try doing it in a way that is at least _theoretically_ backwards compatible so as to not have to wait for a major version, but if this doesn't seem to work let me know and I can tweak further.

I think as long as the existing definitions are not changed and only new lines are added, this should not be a problem. 
My manual additions didn't seem to conflict with the autodiscovery information either.

fkroepfl, 2024-02-19:
After a bit of trial and error, I think the following icons might actually be quite suitable. Any other ideas?

```
Temperature:
    icon: "mdi:thermometer-lines"

Specific Gravity:
    icon: "mdi:trending-down"

Uptime:
    icon: "mdi:timer-sync-outline"

```

[image]

fkroepfl, 2024-02-19:
If the additions are to be made, then it would probably also be advisable to set the "unique_id" as recommended, wouldn't it?

- https://www.home-assistant.io/faq/unique_id/
- https://developers.home-assistant.io/docs/entity_registry_index/

thorrak, 2024-02-19:
> If the additions are to be made, then it would probably also be advisable to set the "unique_id" as recommended, wouldn't it?
> 
> * https://www.home-assistant.io/faq/unique_id/
> * https://developers.home-assistant.io/docs/entity_registry_index/

Yes, but this would definitely be a breaking change, and thus will require a major version increase. Thankfully, I'm likely to do one of those in the not-too-distant future anyways, so I've referenced this in a new issue and will try to release it at the same time as I do #193

PR #252 (2024-02-18): Add additional information to the Home Assistant MQTT Auto Discovery messages
This tags the uptime as being an "additional attribute" (thereby preventing the same reading being sent repeatedly from updating the 'last changed' timestamp), and adds additional device configuration information to the announcement. 

Closes #248
closed 2024-03-03 by thorrak