thorrak/tiltbridge #262: Add support for Bierbot Bricks

trmccart, 2024-05-20:
Bierbot Bricks supports the Tilt but the current recommendation is to use the phone app to send Tilt data. It would be nice if Tiltbridge could handle sending temp and SG data to Bierbot Bricks. 

From a quick look at the data that the Tilt app sends to Bierbot, this appears similar to Brewstatus support.

Feature Request:
-Add support to send Tilt data to the Bierbot Bricks endpoint

thorrak, 2024-05-21:
Hey there - thanks for both the issue and the PR! Although I haven't used Bierbot Bricks myself, from what I know about it, it seems like an interesting project and is definitely something that I would love to help support in TiltBridge!

You caught me at a _slightly_ awkward point in the development cycle, as I'm in the process of working on the next version of TiltBridge, which unfortunately involves changes to sections of the code that this PR touches as well. With that in mind, I'll look to get this merged into a development branch shortly and will seek to get the changes ported into the "next generation" branch I'm currently working on. As soon as those changes are ready, I'll let you know. If you'd be willing to test them out in that branch to make sure I didn't mess anything up in the translation, I would appreciate it.

trmccart, 2024-05-21:
I am actually pretty new to Bierbot also, but it does some cool stuff. It’ll be good to make Tilt data a little easier to use. There is a little follow up work needed at some point to support dynamic backoff for sending data to Bierbot. Maybe it makes sense to wait for the next version as the baseline?  Happy to help. Excited to see what comes next for Tiltbridge!