thorrak/tiltbridge #265: Thousands of DNS requests to tiltbridge.com

aaronpburke, 2024-08-27:
My TiltBridge (running firmware version 1.1.3) is making thousands of requests per minute to tiltbridge.com. I see this in my PiHole logs and the PiHole is rate-limiting the device.

Why does this device need to phone home at all? I do not have TiltBridge Cloud enabled. I currently have no active fermentations (and thus, no TiltBridge devices active/connected). My only targets for streaming are MQTT (to a local Home Assistant on my LAN), Brewers Friend, and GrainFather. None of these should be phoning home to tiltbridge.com.

aaronpburke, 2024-08-27:
[image]
[image]

thorrak, 2024-08-27:
Uhhhhhh.... Weird?

Are you sure it's the TiltBridge device itself that is trying to make those connections, and not another device on your network?

I don't have anything in the code that I can recall that phones home for any reason - the idea of the TiltBridge is that it's as cloud agnostic as possible. There isn't any reference to tiltbridge.com in the source outside of the OTA update code - but that is disabled in every release of TiltBridge to date.

aaronpburke, 2024-08-27:
It's definitely the TB. That's the static IP I've defined for it.

> Pinging tiltbridge.local [192.168.68.137] with 32 bytes of data

I see another reference here: https://github.com/thorrak/tiltbridge/blob/7a5d7e8e017ccb500fa38fad500d7ffc66c4621e/src/parseTarget.cpp#L12

thorrak, 2024-08-27:
Ah, that’s why. That was part of the way that “parse” cloud support had to be built back in the day, but “parse” cloud support has been removed in the latest (unreleased) version I have on my desktop.

Let me see if this code is stable and if I can release a quick patch.

thorrak, 2024-08-28:
Patch is released as v1.2.4 - this should just disable everything (in the back end - I did not want to touch the old front end, no sir) for the parse client.

closed 2024-08-28 by thorrak