thorrak/tiltbridge #275: Display Cycling and Gravity Discrepancy

mstry510, 2025-02-12:
Hello,
I have two issues
1. The display continuously cycles every 10 seconds showing the screen with the readings from the tilt hydrometer and then the tilt bridge logo. Please advise how to stop the cycling and only show the readings.

2. Why is the gravity reading on the TiltBridge display different from the reading in the Tilt app on my phone? 

Any help would be greatly appreciated. Thank you.

thorrak, 2025-02-12:
Hey there - thanks for reaching out!

> The display continuously cycles every 10 seconds showing the screen with the readings from the tilt hydrometer and then the tilt bridge logo. Please advise how to stop the cycling and only show the readings.

This behavior is intentional, and cannot be changed in the assembled firmware by default. The reason that the firmware was built this way is that the readings are frozen behind the scenes when the display is assembled, and are refreshed as the display cycles. Cycling through the Tilt logo is a way to clearly note that the cycle has finished. 

If you would prefer to disable this behavior, you're welcome to - but you will need to compile the firmware yourself. This can be changed by commenting out these lines: https://github.com/thorrak/tiltbridge/blob/7a5d7e8e017ccb500fa38fad500d7ffc66c4621e/src/bridge_lcd.cpp#L420-L423

> Why is the gravity reading on the TiltBridge display different from the reading in the Tilt app on my phone?

There are a few things I can think of that could cause this:

The most obvious reason is that TiltBridge implements specific gravity averaging, where it blends the raw readings off the Tilt into a measured gravity to prevent noise from bubbles, etc. This setting is controlled within the UI which (for the current release) is visible in Settings > General Settings > Specific Gravity Averaging. The default value is 60, but you can set it to 0 if you want to disable gravity averaging. 

Another reason that the readings may seem to differ is that - as I mentioned above - the readings are "frozen" on screen as of the time the screen is rendered, but the Tilt measures roughly once per second. If memory serves, the App reports measurements as they are made, which means that you could be seeing a measurement that is made several seconds after the screen is rendered.

One other thing to be aware of is that TiltBridge (and possibly the Tilt app) allow you to calibrate the readings coming from Tilt. I'm guessing this isn't the case for you, but it's good to be aware that if you have calibrated your Tilts in either place, the calibration is only done on the TiltBridge/App side, and will not affect what the device reports.

In addition to the above, it warrants mentioning that the antenna on your phone is much better than the antenna on the TiltBridge, and it's possible that you are seeing measurements that the TiltBridge is missing. The TiltBridge will cache/report readings that have been made within the past [five minutes](https://github.com/thorrak/tiltbridge/blob/7a5d7e8e017ccb500fa38fad500d7ffc66c4621e/src/tilt/tiltHydrometer.h#L33), so if the signal is weak, you may not have the latest reading.

mstry510, 2025-02-12:
Hello Thorrak,
Great support!!! I really appreciate the quick reply and considerable explanation for each item. That all makes sense. Thank you. 
Best Regards,Sean Lambert

thorrak, 2025-02-12:
Thank you for your interest in the project!

I'm going to close this issue for now, but feel free to continue to comment or reopen if you have any further questions.

closed 2025-02-12 by thorrak