thorrak/tiltbridge #276: Cheap Yellow Display (ESP32-2432S028)

Mtbred, 2025-03-06:
Absolutely love this and it's been working great in a webserver/screenless configuration. I've got it running on a CYD (https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display) without screen functionality. I would assume it would use the TFT firmware but doesn't seem to properly find/configure the display. Any possibility of support for this in the future?

thorrak, 2025-03-06:
I’m about to go to sleep and thus can’t search right now, but can you find out what info if any there is on the display chip? The one I target starts with ILI, but I can’t remember the rest — if it’s that one, you may have some hope

Mtbred, 2025-03-06:
Definitely! Looks like it's an ILI9341 (2.8-inch color TFT display)

thorrak, 2026-03-15:
If it works, that sounds like an ideal solution to me. Better than what I was thinking of, for sure! Also - it might add an opportunity to merge CYD support with TFT support or something similar, if the autodetect model could be extended that far. 

We would definitely want to get @Mtbred to check it out once we have a version merged before we finalize it, though. If we merge in the TFT (LoLin D32 Pro TFT display pinout) support I can test that, but I don't have any CYD variant on hand, unfortunately.