thorrak/tiltbridge #277: New account setup / login issue

sjuhnke, 2025-03-08:
Using the "create account" page for tilt bridge in the web app, I didn't get any emails confirming account setup/verification yesterday and today when I tried again, it said that my account was setup; but then when I try to login, it is invalid combo.  When I click reset password, I never get the email to reset (so seems to be an issue with sending any emails to the gmail.com domain).  Any help?

thorrak, 2025-03-08:
Which web app are you referring to?

sjuhnke, 2025-03-08:
https://tiltbridge.web.app/#/

thorrak, 2025-03-08:
@djmarlow is the one who maintains that, but - as you noted - it seems to be pretty dead, and as such I’ve actually pulled it out of the latest Tilt firmware revisions.

My plan is to implement first party web service functionality as part of Fermentrack.net which is coming soon-ish.

sjuhnke, 2025-03-08:
Got it, my first time trying to setup on an ESP32 and I figured I needed to
flash and then connect the web app to the ESP32, but could be all wrong.

DJMarlow, 2025-03-09:
Sorry guys, been busy with life so I haven't been active lately, but the web app and database should be up and running. I see a user created for sjuhnke, but the email hasn't been verified. It looks like the verification email settings on the database had expired. I've renewed everything. Please recreate your user and a verification email should be sent to you.

sjuhnke, 2025-03-09:
Thanks, that was me - the emails didn’t come through on the Gmail side.
It’s all good.

DJMarlow, 2025-03-10:
I have confirmed that the verification emails are now functioning as designed. Please create a user and confirm that you received verification email and are able to log in and I can close this issue.

sjuhnke, 2025-03-12:
This works now and can be closed, thank you!

thorrak, 2025-06-26:
> Actually I have a follow up comment, howdo we find the GUID on the tiltbridge to attach to your website? I'm using firmware version 2.0.0 beta1.

I created a new issue to track this -- I'll answer there (and close this issue for now, as it's unrelated)

closed 2025-06-26 by thorrak